webdino/amethyst #46: WebViewer の Gecko 78 対応

dynamis, 2021-03-22:
WebViewer 68 をベースに Web Viewer 78 対応を行う

既に一部パッチから順次移植しているが対象バージョンは開発開始時点で最新の 78.7.1 esr をベースとし、5-6 月のリリースまでに更新があれば追従する。

https://github.com/webdino/amethyst/tree/master/FIREFOX_78_7_1esr_RELEASE

dynamis, 2021-04-08:
外部通信系の無効化確認についてメモです:

実際に (プロファイルが無い状態で) 起動して通信の有無をログで確認する:
https://github.com/webdino/amethyst/issues/12#issuecomment-533409939

Firefox が勝手に(？)通信してしまう内容のリスト:
https://support.mozilla.org/ja/kb/how-stop-firefox-making-automatic-connections

NaokiNoguchi, 2021-05-11:
パッチ修正の残件
- removed-files.patch
未対応
- webviewer.patch
about:preferencesが表示されない
原因を解析中

dynamis, 2021-05-11:
removed-files.patch は実施せずとも動作には影響がない。about:preferences が表示されないのは解決したいが必須要件としてサポートすべき機能というわけでもない部分なので、そろそろ実機向けのビルドの準備も並行して始めましょう。

- [ ] [gecko 78 ブランチのレシピ](https://github.com/webdino/meta-browser/tree/firefox-78-wip/dynamic-layers/rust-layer/recipes-browser/firefox) に firefox だけじゃなく webviewer のものを追加する
  - gecko 68 向けのものを参考に https://github.com/webdino/meta-browser/blob/gem-firefox-68.0.0esr/dynamic-layers/rust-layer/recipes-browser/firefox/webviewer_68.0esr.bb
- [ ] docker ビルドで gecko 78 向けのものに webviewer ビルドを追加
  - https://sh171.soum.co.jp/rzg-gecko/rzg-gecko-image
  - webviewer レシピがないから local.conf で外してるのを戻す https://sh171.soum.co.jp/rzg-gecko/rzg-gecko-image/-/commits/ek874-106-esr78/
- [ ] docker image ビルドを量産する
- [ ] 実機試験する
- [ ] ついでにビルド手順を更新しておく #48

dynamis, 2021-06-02:
78.7.1 向けパッチとして作成したが結局 78.10.1 に適用しているので差分は無いとは言え実際にビルド・利用するバージョン向けであることが自明となるようディレクトリ名変えた方が良いかも？
https://github.com/webdino/amethyst/tree/master/FIREFOX_78_7_1esr_RELEASE

NaokiNoguchi, 2021-06-03:
そうですね。ディレクトリ名をFIREFOX_78_10_1esr_RELEASEに変更しておきます。

removed-files.patchは今回どうしましょう。対応したほうが良いでしょうか？

dynamis, 2021-06-03:
removed-files.patch より westone 巻き込む問題の報告が優先ですが、先方がボールもってこちらが待ちになったり余裕ができたら対応しましょう。

最悪そこはナシでもリリースブロックではないはず。

@hATrayflood 私ちゃんと中身みれてないので部分的にでもやらなきゃ的な所はつっこみお願い

NaokiNoguchi, 2021-06-06:
こちらはbrowser要素2つ用意することで解決。
>webviewer.patch
about:preferencesが表示されない
原因を解析中

browser要素を2個にする懸念点として起動時間やメモリの消費量増が考えられる。
実機にて確認
- 起動時間
正確に測ることができていないが、体感で差はなし
- 起動時のメモリ消費量
1つと2つで大きな差は見られない。誤差の範囲内と思われるため問題なし
https://docs.google.com/spreadsheets/d/1oEHbRYhNRhrVZj7VvFy27PvSOYvVxtY89aPMg_MPtO4/edit?usp=sharing

懸念点が問題ないためmasterにマージ
https://github.com/webdino/amethyst/commit/fed1e4e239e228121d2a58adc4b379402e15ced3

dynamis, 2021-12-01:
完了しているのでクローズします

closed 2021-12-01 by dynamis