webdino/amethyst #47: urls.jp から gem プロジェクト用 url に切り替え

dynamis, 2021-03-29:
urls.jp のショート url は個人メンテに依存するサービスなので (ドメインはともかく) Github でリダイレクト定義を書いて Netlify ホストしたサイトをGEM プロジェクト用にホストしたものに切り替える。

- [ ] url 決める
- [ ] サイトデプロイ
- [ ] コードの書き換え

dynamis, 2021-04-06:
urls.jp を netlify ホストにして以下のリダイレクトは復活させておいた。

```
/gemqa  https://github.com/webdino/gecko-embedded/wiki/QA
/gembench  https://github.com/webdino/gecko-embedded/wiki/Benchmark
/gemsample  https://github.com/webdino/gecko-embedded/wiki/Sample-Contents
```

一方で gem プロジェクトドメインっぽくないのでちゃんとリダイレクトサイト別途用意した:

https://r.gecko-embedded.org

https://github.com/dynamis/r.gecko-embedded.org/blob/effd36c860057561ea1bf5645e0bd687cb89202b/_redirects#L4-L10

dynamis, 2021-04-06:
埋め込みコード URL はこちらのように変更お願いします:
https://urls.jp/amethyst  -> https://r.gecko-embedded.org/amethyst

NaokiNoguchi, 2021-04-07:
@dynamis 
念の為の確認ですが、埋め込みコードURLは[browser.startup.homepage](https://github.com/webdino/amethyst/blob/4254e548e08076e2811162663a472261f13a6d9e/FIREFOX_78_7_1esr_RELEASE/webviewer.patch#L283)に設定するURLという認識で間違いないでしょうか？

dynamis, 2021-04-08:
はい。そちらに設定してください。

NaokiNoguchi, 2021-04-08:
小規模な修正でしたので、そのままmasterにpushしました。